ashleyleal/resistor-vending-machine #1: Electrical connections between dispensing units and rest of machine

ashleyleal, 2024-07-12:
## Description

The design of the vending machine is modular, such that dispensing units may be assembled independently and are controlled by their own MCU (Arduino Nano). Connections are required between each dispensing unit and the rest of the machine, including power supply connections and SPI connections with the master MCU (Arduino Mega). The obvious solution is to use wires to make these connections but I think we can come up with something better 😎.

## Task
We need a design for electrical connections between each dispensing unit and the rest of the system that considers the following:
- [ ] **Ease of assembly and maintenance**: the design should allow for quick and easy replacement of dispensing units without disrupting the rest of the system
- [ ] **Stable and reliable connections**: the design should minimize the risk of loose connections
- [ ] **Minimal clutter**: the design should minimize loose wires to keep the system organized 

## Possible Implementations
- Something similar to how GPUs are connected to a motherboard in a PC (PCIe slots)

Thigamore, 2024-07-13:
A possible solution to this would be using I2C though I don't know much about it. Arduino has a library over it and it can connect 128 devices with just two wires coming from supervisor mcu

ashleyleal, 2024-07-21:
[image]

ashleyleal, 2024-07-21:
[image]